raspibolt/raspibolt #1347: /home/bitcoin/.bitcoin permissions vs. /data/bitcoin

lewisphil, 2023-09-18:
### Describe the issue

On the page: https://raspibolt.org/guide/bitcoin/bitcoin-client.html#create-the-bitcoin-user
"Check if the permission cookie can be accessed..."  fails.  The guide says to check this with the following code while logged in as user ADMIN.

`$ls -la /home/bitcoin/.bitcoin/.cookie`

This fails with a permission denied.

`$ls -la /data/bitcoin/.cookie` shows correctly  (  -rw-r----- )
but can a user read a file in another user's home directory, even if that specific file is correctly permissioned?

On the same page under the "Running bitcoind" section, it says to 

" Grant the “bitcoin” group read-permission for the debug log file:"
and then it says the chmod the /data/bitcoin/debug.log.  (notice we are checking /data/bitcoin, not /home/bitcoin/.bitcoin)

This makes sense, other users should be able to access /data/bitcoin, but there's an inconsistency with "other" users accessing bitcoin.  Should they *always* access /data/bitcoin/ and only bitcoind references /home/bitcoin/.bitcoin ?

Then further on the following page: https://raspibolt.org/guide/bitcoin/electrum-server.html
the electrs.conf page lists a daemon_dir as /home/bitcoin/.bitcoin"

when starting the electrs server, it fails with the error:
Error: electrs failed

Caused by:
    0: failed to open bitcoind cookie file: /home/bitcoin/.bitcoin/.cookie
    1: Permission denied (os error 13)

Shouldn't this daemon_dir be /data/bitcoin/  ?

### Location in guide
https://raspibolt.org/guide/bitcoin/bitcoin-client.html#create-the-bitcoin-user

### Environment
 - Hardware platform: Beelink SER5
 - Operating system: Debian 12
 - Version: No LSB modules are available.
Description:	Debian GNU/Linux 12 (bookworm)

### Additional context

BillyMabe456, 2023-09-23:
All I know is I've had issues with these cookie permissions for a few months now, not sure what is going on exactly but something changed after upgrading to bitcoin 25.0 I've been running this node for several years and try to keep up with the changes. My power goes out occasionally or my router freaks out, and my node needs to be restarted. When it does it generates the cookie file with the wrong permissions:  ( -rw------- ) and nothing can access it.

There's code in the bitcoind.service that is supposed to assign the correct permissions to this cookie on startup which doesn't seem to work (probably lacking the correct permissions somehow)
"-startupnotify="chmod g+r /home/bitcoin/.bitcoin/.cookie"

I found if I switch to the bitcoin user, then run "chmod g+r /home/bitcoin/.bitcoin/.cookie" myself, All my RPC connections like electrum or whatever work again. 

Unfortunately, that's as far as my troubleshooting could take me.

jpanachida, 2023-10-30:
As I understand the issue, it is not about .cookie permission but the absolute path of it. Members of bitcoin group (e.g. admin, electrs) should access datadir using /data/bitcoin/ or ~/.bitcoin/ path (if defined). They should not use bitcoin's home directory /home/bitcoin/ because of insufficient permissions.

dosoid, 2023-11-06:
I had the same issue, "Permission denied", even that .cookie file has group "read" access to bitcoin group....change electrs.conf file daemon_dir to /data/bitcoin, now it's working...

closed 2024-02-12 by github-actions[bot]