raspibolt/raspibolt #1363: uvloop installing LNbits

Supersimmons, 2023-10-11:
### Describe the bug
Installing uvloop (0.16.0): Failed

### Location in guide
poetry install --only main

### Screenshots
• Installing uvloop (0.16.0): Failed

  ChefBuildError

  Backend subprocess exited when trying to invoke build_wheel

  running bdist_wheel
  running build
  running build_py
  creating build
  creating build/lib.linux-x86_64-cpython-311
  creating build/lib.linux-x86_64-cpython-311/uvloop
  copying uvloop/_version.py -> build/lib.linux-x86_64-cpython-311/uvloop
  copying uvloop/_noop.py -> build/lib.linux-x86_64-cpython-311/uvloop
  copying uvloop/__init__.py -> build/lib.linux-x86_64-cpython-311/uvloop
  copying uvloop/_testbase.py -> build/lib.linux-x86_64-cpython-311/uvloop
  running egg_info
  writing uvloop.egg-info/PKG-INFO
  writing dependency_links to uvloop.egg-info/dependency_links.txt
  writing requirements to uvloop.egg-info/requires.txt
  writing top-level names to uvloop.egg-info/top_level.txt
  reading manifest file 'uvloop.egg-info/SOURCES.txt'
  reading manifest template 'MANIFEST.in'
  warning: no previously-included files matching '*' found under directory 'vendor/libuv/.git'
  warning: no previously-included files matching '*' found under directory 'vendor/libuv/docs'
  warning: no previously-included files matching '*' found under directory 'vendor/libuv/img'
  adding license file 'LICENSE-APACHE'
  adding license file 'LICENSE-MIT'
  writing manifest file 'uvloop.egg-info/SOURCES.txt'
  /tmp/tmpm_s_6om4/.venv/lib/python3.11/site-packages/setuptools/command/build_py.py:204: _Warning: Package 'uvloop.handles' is absent from the `packages` configuration.
  !!

          ********************************************************************************
          ############################
          # Package would be ignored #
          ############################
          Python recognizes 'uvloop.handles' as an importable package[^1],
          but it is absent from setuptools' `packages` configuration.

          This leads to an ambiguous overall configuration. If you want to distribute this
          package, please make sure that 'uvloop.handles' is explicitly added
          to the `packages` configuration field.

          Alternatively, you can also rely on setuptools' discovery methods
          (for example by using `find_namespace_packages(...)`/`find_namespace:`
          instead of `find_packages(...)`/`find:`).

          You can read more about "package discovery" on setuptools documentation page:

          - https://setuptools.pypa.io/en/latest/userguide/package_discovery.html

          If you don't want 'uvloop.handles' to be distributed and are
          already explicitly excluding 'uvloop.handles' via
          `find_namespace_packages(...)/find_namespace` or `find_packages(...)/find`,
          you can try to use `exclude_package_data`, or `include-package-data=False` in
          combination with a more fine grained `package-data` configuration.

          You can read more about "package data files" on setuptools documentation page:

          - https://setuptools.pypa.io/en/latest/userguide/datafiles.html

          [^1]: For Python, any directory (with suitable naming) can be imported,
                even if it does not contain any `.py` files.
[... 331 lines cut ...]
  creating build/temp.linux-x86_64-cpython-311
  creating build/temp.linux-x86_64-cpython-311/uvloop
  x86_64-linux-gnu-gcc -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/tmp/tmpm_s_6om4/.venv/include -I/usr/include/python3.11 -I/tmp/tmpqtru7x02/uvloop-0.16.0/vendor/libuv/include -c uvloop/loop.c -o build/temp.linux-x86_64-cpython-311/uvloop/loop.o -O2
  uvloop/loop.c:214:12: fatal error: longintrepr.h: No such file or directory
    214 |   #include "longintrepr.h"
        |            ^~~~~~~~~~~~~~~
  compilation terminated.
  error: command '/usr/bin/x86_64-linux-gnu-gcc' failed with exit code 1

  at ~/.local/share/pypoetry/venv/lib/python3.11/site-packages/poetry/installation/chef.py:147 in _prepare
      143│
      144│                 error = ChefBuildError("\n\n".join(message_parts))
      145│
      146│             if error is not None:
    → 147│                 raise error from None
      148│
      149│             return path
      150│
      151│     def _prepare_sdist(self, archive: Path, destination: Path | None = None) -> Path:

Note: This error originates from the build backend, and is likely not a problem with poetry but with uvloop (0.16.0) not supporting PEP 517 builds. You can verify this by running 'pip wheel --use-pep517 "uvloop (==0.16.0)"'.

### Environment
 - Hardware platform: Zimaboard
 - Operating system: Debian GNU/Linux 12 (bookworm)

wankerstar, 2023-10-24:
What version of Python is installed? uvloop was broken for Python 3.12 on release and has seen been patched; not sure if that's the problem but the timing fits.

hellresistor, 2023-12-18:
you need python3.9

maybe this solve your problem https://github.com/raspibolt/raspibolt/pull/1396

closed 2024-02-23 by github-actions[bot]