raspibolt/raspibolt #1375: [BUG] Unable to open cookie file '/home/bitcoin/.bitcoin/.cookie': Permission denied

hellresistor, 2023-11-13:
### Describe the bug
.cookie Permision denied, when run fulcrum
in bitcoin debug.log file returns 
"ThreadRPCServer incorrect password attempt from 127.0.0.1:45744
ThreadRPCServer incorrect password attempt from 127.0.0.1:42374"

### Location in guide
[[Link to the corresponding page and section on <https://raspibolt.org>
](https://raspibolt.org/guide/bonus/bitcoin/fulcrum.html#run-fulcrum)](https://raspibolt.org/guide/bonus/bitcoin/fulcrum.html#run-fulcrum)

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
[If applicable, add screenshots to help explain your problem.](https://prnt.sc/v2ZfenJ9NxLD)

### Environment
 - Raspberry Pi 4
 - Debian 12
 -  Debian GNU/Linux 12 (bookworm)

### Additional context
I have checked and rechecked :|

mzdrowski, 2023-11-13:
Looks like the same issue is being discussed here: https://github.com/raspibolt/raspibolt/issues/1347

I reverted to rpcauth method to get Fulcrum working.

hellresistor, 2023-11-13:
i am using rpcauth on bitcoin.conf :| can you share ?

hellresistor, 2023-11-13:
well solved.

just this /data/fulcrum/fulcrum.conf

# Bitcoin Core settings
bitcoind = 127.0.0.1:8332
rpccookie = /data/bitcoin/.cookie

closed 2023-11-13 by hellresistor