raspibolt/raspibolt #1401: Instructions for Bitcoin Kntos

jesterhodl, 2024-01-03:
### What & why

In order to facilitate users to build nodes according to their expectations, we could provide an optional path of picking Bitcoin Knots - if the user elects to filter spam related to CVE-2023-50428.

> As a user, I want to to be guided in using Bitcoin Core or Knots with respect to vulnerabilities in Bitcoin and its mitigations.

### How

I've written some instructions. Are you open to adding an optional ariant of the "Bitcoin client" chapter with Knots?
https://jesterhodl.com/raspibolt-replacing-bitcoin-core-with-knots/

kristapsk, 2024-01-03:
This have been proposed before (#1215), but nobody has written an actual guide so far.

kristapsk, 2024-01-04:
But that guide also need to mention differences, that can affect users.
* Document `softwareexpiry` config parameter.
* Document that full-RBF is enabled by default (to Core default of allowing only opt-in RBF, `mempoolreplacement=fee,optin` config needs to be added).
* Document that different mempool policies, mainly to filter Inscriptions and big OP_RETURN's, are used by default (which can be disabled with `corepolicy=1` config).

kristapsk, 2024-01-27:
Other Bitcoin node distributions are working on this, adding Knots as alternative to Core, so that user can choose.

Permanode currently gives these options (source - https://twitter.com/parman_the/status/1751075746767073561/photo/1):
* Bitcoin Core v25.0
* Bitcoin Core v26.0
* Bitcoin Core v25.0 w/ FILTER-ORDINALS patch by Luke-Jr (guided compile from sources)
* Bitcoin Core v26.0 w/ FILTER-ORDINALS patch by Luke-Jr (guided compile from sources)
* Bitcoin Knots v25.1.knots20231115 (guided compile from sources)

There is also some support for Knots v25.1 in Start9 - https://twitter.com/BTCtoOblivion/status/1751025548887777436.

closed 2024-04-03 by github-actions[bot]