raspibolt/raspibolt #1414: Stuck at verifying bitcoind operation due to permission problem

bioterra, 2024-03-17:
### Describe the issue
I started with a clean install, and followed all steps in the Guide. When executing:

> $ ls -la /home/bitcoin/.bitcoin/.cookie

I get the response:

> ls: cannot access '/home/bitcoin/.bitcoin/.cookie': Permission denied

### Location in guide
Bitcoin > Verification of bitcoind operations 

### Environment
 - Hardware platform: Raspberry Pi 4
 - Operating system: Debian GNU/Linux 12 (bookworm) 

### Additional context
Also, the subsequent steps in the guide lead errors:

> tail -f /home/bitcoin/.bitcoin/debug.log

Returns:

> tail: cannot open '/home/bitcoin/.bitcoin/debug.log' for reading: Permission denied

And,

> bitcoin-cli getblockchaininfo

Returns:

> error: Could not locate RPC credentials. No authentication cookie could be found, and RPC password is not set.  See -rpcpassword and -stdinrpcpass.  Configuration file: (/home/admin/.bitcoin/bitcoin.conf)

Probably all related. What to do?

kristapsk, 2024-03-17:
User from which you are running these commands must belong to the `bitcoin` group.

bioterra, 2024-03-17:
Thanks, I am running this from the "admin" user, as per the guide, and I added "admin" to the bitcoin group via: 

> sudo adduser admin bitcoin

(If I re-run the above command it says: "adduser: The user `admin' is already a member of `bitcoin'.")

bioterra, 2024-03-18:
I found a solution, I had to give read+execute rights to members of the `bitcoin` group on the folder `/home/bitcoin/`

> sudo chmod g+rx /home/bitcoin/

After that, the command ran as expected from the user `admin`:

>admin@raspibolt:~ $ ls -la /home/bitcoin/.bitcoin/.cookie
>-rw-r----- 1 bitcoin bitcoin 75 Mar 17 20:36 /home/bitcoin/.bitcoin/.cookie

I do think this `chmod` command is not part of the guide/walkthrough, should it be added?

closed 2024-03-18 by bioterra