raspibolt/raspibolt #1415: Unable to upgrade Electrum Server to v0.10.4

x0utlier, 2024-03-25:
I am attempting to upgrage Electrum Server as per the instructions in the section:
	Bitcoin | Electrum Server | For the future: Electrs upgrade
	https://raspibolt.org/guide/bitcoin/electrum-server.html#for-the-future-electrs-upgrade

Currently running Electrum Server v0.9.13 and upgrading to v0.10.4

System is a Raspberry Pi 4B running Debian GNU/Linux 11 (bullseye)

When I get to this part of the instructions:
```
$ cargo clean
```
I get this error:
```
error: failed to parse manifest at `/home/admin/rust/electrs/Cargo.toml`

Caused by:
failed to parse the `edition` key

Caused by:
this version of Cargo is older than the `2021` edition, and only supports `2015` and `2018` editions.
```
Cargo version is v1.46.0, it was installed as per the Raspibolt guide, I followed the instructions precisely.

I have run
```
$ sudo apt-get update && sudo apt-get upgrade
```
and rebooted the system, no luck.

Any help would be appreciated, thanks!

fots1110, 2024-04-26:
I have the same issue.
The Bonus section has the option to install `Eectrum Personal Server ` as an alternative to electrs. However there is no instructions of how to uninstall electrs, Which has a large database.

llubel, 2024-04-28:
Same issue here.

CorLeoGit, 2024-05-12:
I have just started with raspibolt. At least the blockchain synced now completly and  wanna go further with the installation process. Stuck with the same issue. 

Command "$ cargo build --locked --release" end up with:
___
error: failed to parse manifest at `/home/admin/rust/electrs/Cargo.toml`

Caused by:
  failed to parse the `edition` key

Caused by:
  this version of Cargo is older than the `2021` edition, and only supports `2015` and `2018` editions.
___

I looked a little arround in google to find a solution. 
1- dist upgrade to debian 12 
2- installing "rustup --> https://rustup.rs/"
3- using Eectrum Personal Server instead "electrs".

Of course, the main goal for a node is the security and reliability of the raspibolt. With which solution you sould I go? What is your recommendation?

managai, 2024-06-10:
I solved installing dependencies in this way:

```bash
# install clang and cmake
admin@raspibolt:~/rust/electrs $ sudo apt install clang cmake
```

```bash
# install latest cargo release
admin@raspibolt:~/rust/electrs $ curl https://sh.rustup.rs -sSf | sh
```

```bash
# reload $PATH
admin@raspibolt:~/rust/electrs $ . "$HOME/.cargo/env"
```

```bash
# first time sync latest cargo components
admin@raspibolt:~/rust/electrs $ cargo --version
```

now  I can install latest electrs release
```bash
# latest electrs release
admin@raspibolt:~/rust/electrs $ VERSION="0.10.5"
```

above commands might replace ```sudo apt install cargo clang cmake```  in https://raspibolt.org/guide/bitcoin/electrum-server.html#install-dependencies

x0utlier, 2024-06-11:
Thank you @managai, problem solved.

CorLeoGit, 2024-07-10:
Thanks alot @managai . This solves the problem.

closed 2024-09-15 by github-actions[bot]