raspibolt/raspibolt #1416: LNDg: 502 Bad Gateway error even after adding user 'www-data' to group 'lndg'

tomesink, 2024-03-25:
### Describe the issue
I am getting exactly the same errors as in #1125 . I have followed the guide step by step and the user `www-data` is added into the `lndg` group which was the solution for the 1125 (and now it is a part of the guide).

I really have no idea what's going on. I'll appreciate any help.

### Location in guide
[Here](https://github.com/raspibolt/raspibolt/blob/21b9ad04ba5cec36e62c29043c5988004fb549c4/guide/bonus/lightning/lndg.md?plain=1#L459)

### Environment
 - Hardware Raspberry PI 4B
 - Operating system: `Raspbian (bookworm) 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux`

### Additional context
`cat /etc/group` :  
```sh
www-data:x:33:lndg
lndg:x:1012:www-data
```

`sudo systemctl status nginx`:  
```sh
● nginx.service - A high performance web server and a reverse proxy server
     Loaded: loaded (/lib/systemd/system/nginx.service; enabled; preset: enabled)
     Active: active (running) since Mon 2024-03-25 20:21:19 CET; 7s ago
       Docs: man:nginx(8)
    Process: 1679 ExecStartPre=/usr/sbin/nginx -t -q -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
    Process: 1680 ExecStart=/usr/sbin/nginx -g daemon on; master_process on; (code=exited, status=0/SUCCESS)
   Main PID: 1681 (nginx)
      Tasks: 5 (limit: 3916)
        CPU: 204ms
     CGroup: /system.slice/nginx.service
             ├─1681 "nginx: master process /usr/sbin/nginx -g daemon on; master_process on;"
             ├─1682 "nginx: worker process"
             ├─1683 "nginx: worker process"
             ├─1684 "nginx: worker process"
             └─1685 "nginx: worker process"
```

`sudo journalctl -f -u uwsgi`:  

```sh
Mar 25 19:05:13 raspibolt systemd[1]: Starting uwsgi.service - LNDg uWSGI app...
Mar 25 19:05:13 raspibolt uwsgi[1292]: [uWSGI] getting INI configuration from /home/lndg/lndg/lndg.ini
Mar 25 19:05:16 raspibolt systemd[1]: Started uwsgi.service - LNDg uWSGI app.
Mar 25 20:10:18 raspibolt systemd[1]: Stopping uwsgi.service - LNDg uWSGI app...
Mar 25 20:10:18 raspibolt systemd[1]: uwsgi.service: Deactivated successfully.
Mar 25 20:10:18 raspibolt systemd[1]: Stopped uwsgi.service - LNDg uWSGI app.
Mar 25 20:10:18 raspibolt systemd[1]: uwsgi.service: Consumed 2.056s CPU time.
Mar 25 20:10:18 raspibolt systemd[1]: Starting uwsgi.service - LNDg uWSGI app...
Mar 25 20:10:19 raspibolt uwsgi[1547]: [uWSGI] getting INI configuration from /home/lndg/lndg/lndg.ini
Mar 25 20:10:21 raspibolt systemd[1]: Started uwsgi.service - LNDg uWSGI app.

```

` cat /var/log/nginx/error.log`:   
```sh
2024/03/25 20:23:00 [crit] 1709#1709: *4 connect() to unix:///home/lndg/lndg/lndg.sock failed (13: Permission denied) while connecting to upstream, client: X.X.X.X, server: _, request: "GET /favicon.ico HTTP/1.1", upstream: "uwsgi://unix:///home/lndg/lndg/lndg.sock:", host: "raspibolt.local:8889", referrer: "https://raspibolt.local:8889/"
```

warioishere, 2024-04-05:
same here. everything is setup appropriate. No way to access gui. I even set 777 for lndg.sock to test

warioishere, 2024-04-05:
the nginx Konfiguration from the nginx. sh script doesnt work. A friend gave me another one:

```
upstream django {
    server unix:///home/lndg/lndg/lndg.sock; # for a file socket
}

server {
    # the port your site will be served on, use port 80 unless setting up ssl certs, then 443
    listen 8889 ssl;
    listen [::]:8889 ssl;
    ssl_certificate /etc/ssl/certs/nginx-selfsigned.crt;
    ssl_certificate_key /etc/ssl/private/nginx-selfsigned.key;
    ssl_session_timeout 4h;
    ssl_protocols TLSv1.3;
    ssl_prefer_server_ciphers on;

    server_name _; # you can substitute your node IP address or a custom domain like lndg.local (just make sure to update your local hosts file)
    charset     utf-8;

    # max upload size
    client_max_body_size 75M;   # adjust to taste

    # max wait for django time
    proxy_read_timeout 180;

    # Django media
    location /static {
        alias /home/lndg/lndg/gui/static; # your Django project's static files - amend as required
    }

    # Finally, send all non-media requests to the Django server.
    location / {
        uwsgi_pass  django;
        include     /home/lndg/lndg/uwsgi_params; # the uwsgi_params file
    }
}
```

this made it to work

tomesink, 2024-04-06:
> the nginx Konfiguration from the nginx. sh script doesnt work. A friend gave me another one:
>
> ```
> upstream django {
>     server unix:///home/lndg/lndg/lndg.sock; # for a file socket
> }
>
> server {
>     # [...]

this is the `/etc/nginx/sites-available/lndg-ssl.conf` and it is exactly the same as in the guide... didn't you copy&pasted wrong text by a chance? There is no difference between the guide and config you've provided... if yes pls point me to the difference.

closed 2024-06-12 by github-actions[bot]

LeoSpyke, 2024-10-09:
Were you able to solve the problem? @tomesink @warioishere