raspibolt/raspibolt #1427: [FEATURE REQ] Harddisk upgrade? 1TB SSD getting full

uskrusader, 2024-05-07:
Running raspibolt 3 now for a couple of years with a 1TB SSD.
disksize is approaching 6.48GB.

How difficult will it be to replace or add another SSD when my 1TB disk is full?
Guide?

thanks!

on4r, 2024-05-07:
Yeah, many will have to upgrade sooner or later. 
Did you take a look at https://raspibolt.org/guide/bonus/raspberry-pi/upgrade-external-drive.html?

on4r, 2024-05-09:
If you use a external drive to boot (that is, without a SD-card), you can just clone your old drive to a new one with [clonezilla](https://clonezilla.org/). Make sure to use the `-k1` option to extend the partition table proportionally. Just upgraded yesterday from 1TB to 2TB (SSD). Worked flawlessly 👌

lip2443, 2024-06-18:
> If you use a external drive to boot (that is, without a SD-card), you can just clone your old drive to a new one with [clonezilla](https://clonezilla.org/). Make sure to use the `-k1` option to [...]

A tutorial of how to perform that would be great

closed 2024-08-24 by github-actions[bot]