raspibolt/raspibolt #1428: Installation process of Homer Raspibolt Dashboard

CorLeoGit, 2024-05-08:
### Describe the issue
Try to install the homer dashboard like suggested here: https://raspibolt.org/guide/bonus/raspberry-pi/homer.html#homer

But get warning during executing "npm install" --> see attachment "npm_install_warning.txt"
Get Error during executing "npm run build" --> see screenshot "npm_run_build.png"

### Location in guide
Link to the corresponding page and section on <https://raspibolt.org>
https://raspibolt.org/guide/bonus/raspberry-pi/homer.html#homer

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: [e.g. Raspberry Pi 4, 8GB]
 - Operating system: [Raspberry Pi OS (Legacy, 64 bit) Lite]
 - Version: Debian GNU/Linux 11 (bullseye)

I installed my Pi stricty holding your installation guidline (3 days ago) used the bitcoin core "bitcoin-27.0-aarch64-linux-gnu.tar.gz". Softwae update performed.
[npm_install_warning.txt](https://github.com/raspibolt/raspibolt/files/15247504/npm_install_warning.txt)
[image]
.

### Additional context
Add any other context about the problem here.

kristapsk, 2024-05-08:
Don't have experience with Homer, but you should upgrade your nodejs version. It's already required by lightning webapp (RTL) guide - https://raspibolt.org/guide/lightning/web-app.html.

CorLeoGit, 2024-05-08:
Wau. Thanks for the fast reply. Exactly this was the issue. Problem fixed. Dashboard up and runnning. Thanks :)

closed 2024-05-08 by CorLeoGit