raspibolt/raspibolt #1445: Nodejs 16 is no longer supported

qlrd, 2024-08-22:
### Describe the issue

After following a tutorial to install JoinMarket, received this message:

```bash
 Node.js 16.x is no longer actively supported!

  You will not receive security or critical stability updates for this version.

  You should migrate to a supported version of Node.js as soon as possible.
  Use the installation script that corresponds to the version of Node.js you
  wish to install. e.g.

   * https://deb.nodesource.com/setup_16.x — Node.js 16 "Gallium" (deprecated)
   * https://deb.nodesource.com/setup_18.x — Node.js 18 "Hydrogen" (Maintenance)
   * https://deb.nodesource.com/setup_19.x — Node.js 19 "Nineteen" (deprecated)
   * https://deb.nodesource.com/setup_20.x — Node.js 20 LTS "Iron" (recommended)
   * https://deb.nodesource.com/setup_21.x — Node.js 21 "Iron" (current)

  Please see https://github.com/nodejs/Release for details about which
  version may be appropriate for you.

  The NodeSource Node.js distributions repository contains
  information both about supported versions of Node.js and supported Linux
  distributions. To learn more about usage, see the repository:
   https://github.com/nodesource/distributions

================================================================================
================================================================================
```
Since many applications use nodejs, maybe it's worth to update the guide to another nodejs version?

### Location in guide
https://raspibolt.org/guide/bonus/bitcoin/joinmarket.html

### Environment
 - Hardware platform: OrangePi 5
 - Operating system: Armbian
 - Version: Ubuntu 22.04.4 LTS

qlrd, 2024-10-22:
There's a fix for this issue?

kristapsk, 2024-10-31:
> https://raspibolt.org/guide/bonus/bitcoin/joinmarket.html

JoinMarket guide definitely doesn't need NodeJS. Guess you meant Jam guide (JoinMarket web UI).

Not sure what NodeJS versions Jam supports, it would be best to look at https://github.com/joinmarket-webui/jam docs.

Maybe @theborakompanioni can comment.

theborakompanioni, 2024-10-31:
> > https://raspibolt.org/guide/bonus/bitcoin/joinmarket.html
> 
> JoinMarket guide definitely doesn't need NodeJS. Guess you meant Jam guide (JoinMarket web UI).

Probably meant https://raspibolt.org/guide/bonus/bitcoin/Jam.html#nodejs.
There is a guide linked in the article on how to update nodejs.

> Not sure what NodeJS versions Jam supports, it would be best to look at https://github.com/joinmarket-webui/jam docs.

You should be able to build Jam with node >=16.11.0. Docker images used v20.10.0 for building up until Jam v0.3.0.
Future versions [will require >=v22.11.0](https://github.com/joinmarket-webui/jam/pull/866).

closed 2025-01-06 by github-actions[bot]