raspibolt/raspibolt #1462: [BUG] bitcoin.conf

MeinOfenBrennt, 2024-12-24:
### Describe the bug
in file:bitcoin.conf --> https://raspibolt.org/guide/bitcoin/bitcoin-client.html
in line
rpcauth=<replace with your own auth line generated by rpcauth.py>
when i replace the string bitcoind doesn't start from systemd. 
it shows permission problems.

sudo grep -i bitc /var/log/syslog
Dec 24 19:14:11 BTC-Node bitcoind[2568]: Error: Settings file could not be read:
Dec 24 19:14:11 BTC-Node bitcoind[2568]: - /home/bitcoin/.bitcoin/settings.json. Please check permissions.

bitcoind show:
2024-12-24T18:32:21Z Using rpcauth authentication.
2024-12-24T18:32:21Z Invalid -rpcauth argument.

**solution:**
you have to write user:hashed-password
rpcauth=**rpcuser**:<replace with your own auth line generated by rpcauth.py>

then all works fine.
took me some hours :-(
but great guidance

br
Arnold

JhovanEscobidal, 2024-12-27:
…

lustremedia, 2025-01-15:
So it looks like your issue has been resolved. You could close it then ...

closed 2025-03-23 by github-actions[bot]