raspibolt/raspibolt #1464: [BUG]

JhovanEscobidal, 2024-12-27:
### Describe the bug
A clear and concise description of what the bug is.

### Location in guide
Link to the corresponding page and section on <https://raspibolt.org>

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Environment
 - Hardware platform: [e.g. Raspberry Pi 4, Odroid, ... ]
 - Operating system: [e.g, Raspberry Pi OS 64bit, Debian 11, ... ]
 - Version: [ output of `lsb_release -d` ]

### Additional context
Add any other context about the problem here.

JhovanEscobidal, 2024-12-27:
…

closed 2024-12-27 by kristapsk

saubyk, 2024-12-28:
this user should be blocked for creating spam tickets

kristapsk, 2024-12-29:
> this user should be blocked for creating spam tickets

I already reported to GitHub. Don't have necessary permissions to ban him from organization. Likely @Stadicus can do it.

Stadicus, 2025-01-03:
Done ✅