raspibolt/raspibolt #1475: Please add a guide for how to upgrade debian version

soc7, 2025-02-21:
I did a magration from raspiOS bullseye to bookworm this week and it was very fast and smooth, it was as simply as changing the sources.list and sources.list.d as any other debian migration and everything works like before bitcoin core, electrs, LND litd and other services I have running.

gwilkinson01, 2025-04-26:
It could be useful for the future for those who built their node on Bullseye. 

One connected issue is that the install guide also needs updating. Two options here:

1. Update /Raspberry Pi/Operating System to instruct users to install Bookworm instead of Bullseye. May need to further test that there are no issues with installing/running bitcoind , lnd , electrs etc. on this version.
2. Continue to instruct new users to use Bullseye, but then you need to adjust the guidance in Electrs since the package installer does not install the latest version of Cargo needed for the build of Electrs,. For that you need to install Cargo via Rustup.

soc7, 2025-04-26:
What about those users running a node on bullseye? Are you suggesting to close all channels and build a new node? I simply upgraded my bullseye node to bookworm, doing the necessary adjustments and I have all running fine including electrs, bitcoin core, LND and all my services. I needed to upgrade the kernel manually as it's from raspiOS not debian directly, but I have 6.12.20 and everything works perfectly. I could help with a guide for those willing to do the same.

gwilkinson01, 2025-04-27:
No I agree that a bonus guide describing how to upgrade from Bullseye is a good idea. 

I was also suggesting the main guide needs some updating, but that this is a seperate issue. Sorry for the confusion!

martinatime, 2025-06-02:
I'm starting from scratch using bookworm. I think the guide should be updated to use bookworm and also have a separate "bonus" guide for upgrading from bullseye to bookworm.

soc7, 2025-06-03:
> I'm starting from scratch using bookworm. I think the guide should be updated to use bookworm and also have a separate "bonus" guide for upgrading from bullseye to bookworm.

Yes, especially now that debian trixie is coming out bookworm to trixie would also be very helpful, anyway I will upgrade as soon is trixie is out, but I know some people scared of upgrading and still using bullseye.

closed 2025-08-09 by github-actions[bot]

thool2b3, 2025-09-07:
> What about those users running a node on bullseye? Are you suggesting to close all channels and build a new node? I simply upgraded my bullseye node to bookworm, doing the necessary adjustments and [...]

Hi @ghost, 
I would like to see a guide how to upgrade from bullseye to bookworm.
Thanks a lot.

thool2b3, 2025-09-07:
> I'm starting from scratch using bookworm. I think the guide should be updated to use bookworm and also have a separate "bonus" guide for upgrading from bullseye to bookworm.

Hi @martinatime ,
Could you tell whether this guide been written or is in the making? 
I am a newbie with linux and desperate for some guide. 
At the moment I experience an extreme slow IBD. Strange is that the IBD was very quick until about block 800000.
No clue why that is so slow now but on my first attempt when I used RaspiOS based on bulleye the whol IBD was finished within 4 days. Now I consider starting all over again installing bullseye again do the IBD and then upgrading to bookworm.
Not sure if that makes sense. So I would be grateful for any advice. 
Thanks

thool2b3, 2025-09-09:
>
> I don't know if the guide was written but I was starting from scratch on new hardware and therefore was going with the latest stable OS. The IBD took almost 3 months for me [bitcoin/bitcoin#32455 (comment)](https://github.com/bitcoin/bitcoin/issues/32455#issuecomment-3221920491) Now that it is done everything seems to be fine but that definitely took WAY too long. It might be worth it to run the IBD on Bullseye....and then upgrade to Bookworm. Check out https://forums.raspberrypi.com/viewtopic.php?t=385353

Hi @martinatime, thank you for your answer. IBD for three month this is crazy and weird. Even if I would wait as long as three month for the IBD I would still think that something is wrong and this would make me at least a bit uneasy. 
I have installed the original raspibolt installation with the recommended RaspiOS based on debian 11 again and the IBD was done within four (4) days. Now I could copy the Blockchain data over to the new installation with RaspiOS based on debian 12 but I would still have this uneasy feeling. 
So I think to keep the RaspiBolt project alive some people with the proper knowledge should seriously investigate the problem.
I would offer to share my notes and experiences if it is helpful.

thool2b3, 2025-09-10:
@martinatime Icould successfully copy the Blockchain data over to my new 2TB SSD with RaspiBolt on debian 12. The process was quite smooth. Now I can go ahead and try installing the latest version of electrs which failed on debian 11.