raspibolt/raspibolt #1476: electrs failed to start

Merlin673, 2025-03-08:
**### Describe the issue**
I am following the setup instructions. Bitcoin core is installed and synced, I have installed electrs and am trying to start it for the first time to check it is functioning.

**### Location in guide**
I am in the Bitcoin section and Electrum server at line: 

electrs --conf /data/electrs/electrs.conf

**### Screenshots**
This is the console output:

electrs@raspberrypi:~ $ electrs --conf /data/electrs/electrs.conf
Starting electrs 0.9.14 on aarch64 linux with Config { network: Bitcoin, db_path: "/data/electrs/db/bitcoin", daemon_dir: "/data/bitcoin", daemon_auth: CookieFile("/data/bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, skip_block_download_wait: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs 0.9.14 (Electrum Rust Server)!", signet_magic: f9beb4d9, args: [] }
[2025-03-08T12:12:36.685Z DEBUG tiny_http] Server listening on 127.0.0.1:4224
[2025-03-08T12:12:36.685Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
[2025-03-08T12:12:36.685Z INFO  electrs::server] serving Electrum RPC on 127.0.0.1:50001
[2025-03-08T12:12:36.685Z DEBUG tiny_http] Running accept thread
[2025-03-08T12:12:36.743Z INFO  electrs::db] "/data/electrs/db/bitcoin": 237 SST files, 0.000190785 GB, 0.000000021 Grows
[2025-03-08T12:12:36.743Z DEBUG electrs::db] DB Some(Config { compacted: false, format: 0 })
[2025-03-08T12:12:36.752Z DEBUG bitcoincore_rpc] JSON-RPC request: getblockchaininfo []
[2025-03-08T12:12:36.753Z DEBUG bitcoincore_rpc] JSON-RPC request: getnetworkinfo []
[2025-03-08T12:12:36.754Z INFO  electrs::db] closing DB at /data/electrs/db/bitcoin
Error: electrs failed

Caused by:
    0: bitcoind RPC polling failed
    1: daemon not available
    2: JSON error: invalid type: sequence, expected a string

**### Environment**
Pi 5
Pi OS 64 bit
Debian GNU/Linux 12 (bookworm)

**### Additional context**
Add any other context about the problem here.

bitcoind is running:

systemctl status bitcoind
* bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; preset: enabled)
     Active: active (running) since Wed 2025-03-05 23:20:28 GMT; 2 days ago
   Main PID: 49620 (bitcoind)
      Tasks: 18 (limit: 9247)
        CPU: 17h 37min 20.035s
     CGroup: /system.slice/bitcoind.service
             `-49620 /usr/local/bin/bitcoind -daemon -pid=/run/bitcoind/bitcoind.pid -conf=/home/bitcoin/.bitcoin/bitcoin.conf -datadir=/home/bitcoin/.bitcoin "-startupnotify=chmod g+r /home/bitcoin/.bitcoin/.cookie"

Mar 05 23:20:28 raspberrypi systemd[1]: Starting bitcoind.service - Bitcoin daemon...
Mar 05 23:20:28 raspberrypi bitcoind[49619]: Bitcoin Core starting
Mar 05 23:20:28 raspberrypi systemd[1]: Started bitcoind.service - Bitcoin daemon.

I can query bitcoind via the cli tool as the electrs user:

electrs@raspberrypi:/home/root2 $ bitcoin-cli getblockchaininfo
{
  "chain": "main",
  "blocks": 886858,
  "headers": 886858,
  "bestblockhash": "00000000000000000001a4b0a704457e12191036d6499a18bd0b0e9ee1239ec5",
  "difficulty": 110568428300952.7,
  "time": 1741437120,
<snip>

I am curious if in the electrs debug log output there should be anything between the square brackets at the end of this line:

[2025-03-08T12:12:36.752Z DEBUG bitcoincore_rpc] JSON-RPC request: getblockchaininfo []

Of the three root causes that the electrs output suggests:

    0: bitcoind RPC polling failed _- we can query getblockchaininfo as the user electrs_
    1: daemon not available _- bitcoind is service is running_
    2: JSON error: invalid type: sequence, expected a string _- is this the problem_

Can you suggest what I need to check / do next? I have already set logging a debug level but I don't think it has provided much more to point me in the direction of what to do next.

Please help, thanks in advance.

Merlin673, 2025-03-09:
After enabling debug mode TRACE we have some more info:

root@raspberrypi:/home/root2# nano /data/electrs/electrs.conf
root@raspberrypi:/home/root2# electrs --conf /data/electrs/electrs.conf
Starting electrs 0.9.14 on aarch64 linux with Config { network: Bitcoin, db_path: "/data/electrs/db/bitcoin", daemon_dir: "/data/bitcoin", daemon_auth: CookieFile("/data/bitcoin/.cookie"), daemon_rpc_addr: 127.0.0.1:8332, daemon_p2p_addr: 127.0.0.1:8333, electrum_rpc_addr: 127.0.0.1:50001, monitoring_addr: 127.0.0.1:4224, wait_duration: 10s, jsonrpc_timeout: 15s, index_batch_size: 10, index_lookup_limit: None, reindex_last_blocks: 0, auto_reindex: true, ignore_mempool: false, sync_once: false, skip_block_download_wait: false, disable_electrum_rpc: false, server_banner: "Welcome to electrs 0.9.14 (Electrum Rust Server)!", signet_magic: f9beb4d9, args: [] }
[2025-03-09T10:31:14.211Z DEBUG tiny_http] Server listening on 127.0.0.1:4224
[2025-03-09T10:31:14.211Z INFO  electrs::metrics::metrics_impl] serving Prometheus metrics on 127.0.0.1:4224
[2025-03-09T10:31:14.211Z INFO  electrs::server] serving Electrum RPC on 127.0.0.1:50001
[2025-03-09T10:31:14.212Z DEBUG tiny_http] Running accept thread
[2025-03-09T10:31:14.316Z INFO  electrs::db] "/data/electrs/db/bitcoin": 2893 SST files, 0.002328865 GB, 0.000000021 Grows
[2025-03-09T10:31:14.316Z DEBUG electrs::db] DB Some(Config { compacted: false, format: 0 })
[2025-03-09T10:31:14.328Z DEBUG bitcoincore_rpc] JSON-RPC request: getblockchaininfo []
[2025-03-09T10:31:14.328Z TRACE bitcoincore_rpc] JSON-RPC response for getblockchaininfo: {"chain":"main","blocks":887012,"headers":887012,"bestblockhash":"00000000000000000000eb5cbd76e26c1f87ffb88607812f2d7dfbf8840872ca","difficulty":110568428300952.7,"time":1741514858,"mediantime":1741513794,"verificationprogress":0.9999918217661166,"initialblockdownload":false,"chainwork":"0000000000000000000000000000000000000000b343968a6ee07a2c9acc1986","size_on_disk":732082085405,"pruned":false,"warnings":[]}
[2025-03-09T10:31:14.328Z DEBUG bitcoincore_rpc] JSON-RPC request: getnetworkinfo []
[2025-03-09T10:31:14.328Z TRACE bitcoincore_rpc] JSON-RPC response for getnetworkinfo: {"version":280100,"subversion":"/Satoshi:28.1.0/","protocolversion":70016,"localservices":"0000000000000c09","localservicesnames":["NETWORK","WITNESS","NETWORK_LIMITED","P2P_V2"],"localrelay":true,"timeoffset":0,"networkactive":true,"connections":10,"connections_in":0,"connections_out":10,"networks":[{"name":"ipv4","limited":false,"reachable":true,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"ipv6","limited":false,"reachable":true,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"onion","limited":false,"reachable":true,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true},{"name":"i2p","limited":true,"reachable":false,"proxy":"","proxy_randomize_credentials":false},{"name":"cjdns","limited":true,"reachable":false,"proxy":"127.0.0.1:9050","proxy_randomize_credentials":true}],"relayfee":0.00001000,"incrementalfee":0.00001000,"localaddresses":[],"warnings":[]}
[2025-03-09T10:31:14.328Z INFO  electrs::db] closing DB at /data/electrs/db/bitcoin
Error: electrs failed

Caused by:
    0: bitcoind RPC polling failed
    1: daemon not available
    2: JSON error: invalid type: sequence, expected a string

This makes me conclude:
1> it is not an RPC auth issue at electrs can clearly query bitcoin
2> it is not a bitcoin daemon issue as it it was then we wouldn't be able to query it in the first place
3> the JSON looks ok to me

I have also tried, changing ufw default policy to permit all incoming and it has made no difference.

I have checked user group membership and this looks ok to me.

Anyone have any ideas?

Merlin673, 2025-03-09:
ok, done some more digging. At the end of this file:

root@raspberrypi:/data/electrs/db/bitcoin# cat LOG

I have:

** Compaction Stats [spending] **
Priority    Files   Size     Score Read(GB)  Rn(GB) Rnp1(GB) Write(GB) Wnew(GB) Moved(GB) W-Amp Rd(MB/s) Wr(MB/s) Comp(sec) CompMergeCPU(sec) Comp(cnt) Avg(sec) KeyIn KeyDrop
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Uptime(secs): 0.1 total, 0.0 interval
Flush(GB): cumulative 0.000, interval 0.000
AddFile(GB): cumulative 0.000, interval 0.000
AddFile(Total Files): cumulative 0, interval 0
AddFile(L0 Files): cumulative 0, interval 0
AddFile(Keys): cumulative 0, interval 0
Cumulative compaction: 0.00 GB write, 0.00 MB/s write, 0.00 GB read, 0.00 MB/s read, 0.0 seconds
Interval compaction: 0.00 GB write, 0.00 MB/s write, 0.00 GB read, 0.00 MB/s read, 0.0 seconds
Stalls(count): 0 level0_slowdown, 0 level0_slowdown_with_compaction, 0 level0_numfiles, 0 level0_numfiles_with_compaction, 0 stop for pending_compaction_bytes, 0 slowdown for pending_compaction_bytes, 0 memtable_compaction, 0 memtable_slowdown, interval 0 total count

** File Read Latency Histogram By Level [spending] **
2025/03/09-13:44:55.180089 7fff32f21000 [db/db_impl/db_impl.cc:435] Shutdown: canceling all background work
2025/03/09-13:44:55.180439 7fff32f21000 [db/db_impl/db_impl.cc:612] Shutdown complete

This appears to be where the shutdown is getting initiated from, google searches have not yielded anything useful.

Thoughts anyone?

Merlin673, 2025-03-10:
upgraded electrs to 0.10.9 from 0.9.14 appears to have fixed the above problems, it is currently indexing. Had to remove the timestamps line from the electrs.conf to get service to start as option depreciated.

raspibolt2025, 2025-03-21:
@Merlin673 I tried upgrading electrs to 0.10.9 but got 
```
admin@raspibolt:~/rust/electrs $ cargo build --locked --release
error: failed to parse manifest at `/home/admin/rust/electrs/Cargo.toml`

Caused by:
  failed to parse the `edition` key

Caused by:
  this version of Cargo is older than the `2021` edition, and only supports `2015` and `2018` editions.
```

did that happen to you too?

zynos, 2025-03-29:
It failed for me too but because it waited for bitcoind.service startup to finish so I replaced
`-startupnotify="chmod g+r /home/bitcoin/.bitcoin/.cookie"`
with this
`-startupnotify="chmod g+r /home/bitcoin/.bitcoin/.cookie && systemd-notify --ready"`
in 
`/etc/systemd/system/bitcoind.service`

closed 2025-06-04 by github-actions[bot]