raspibolt/raspibolt #1479: Update GPG signature verification link for Jam

bsin0, 2025-04-05:
### Describe the issue
The guide states to use dergigi for pgp signature verification which does not work for verifying Jam version 3.0 onwards which only applies to versions before 3.0. 
"curl https://dergigi.com/PGP.txt | gpg --import"

On Jamdocs.org, for Jam version 3.0 onwards it states to use:
"curl https://raw.githubusercontent.com/joinmarket-webui/jam-docker/refs/heads/master/standalone/pubkeys/tbk.asc | gpg --import"

### Location in guide
Under "Install Jam" instructions (https://raspibolt.org/guide/bonus/bitcoin/Jam.html) where it states:
"Obtain developer pgp signature for verification"

### Screenshots
N/A

### Environment
 - Hardware platform: N/A
 - Operating system: N/A
 - Version: N/A

### Additional context
Updated verification instructions found here
https://jamdocs.org/software/verification/

gwilkinson01, 2025-04-17:
My understanding here is that v0.3.0 has not been thoroughly tested with the rest of the raspibolt configuration, hence why in the Jam guide under "Retrieve Source Code" users are instructed to retrieve version 0.1.5.0: `$ VERSION="v0.1.5"`.

Saying that I tested in my environment (Pi4, Pi OS 64 bit, Debian GNU/Linux 12 (bookworm), Bitcoind v29.0) and don't seem to have experienced any problems running Jam so far.

gwilkinson01, 2025-04-26:
pr was merged. can be closed.