raspibolt/raspibolt #1481: Large SSD, Get an insufficent space error while unpacking bitcoin core client

pleb4868, 2025-04-23:
### Describe the issue
I'm following the guide on a 4TB SSD and at the stage where I'm unpacking the bitcoin core .tar i get an insufficient space error

### Location in guide
[link to relevant place in guide <https://raspibolt.org/guide/bitcoin/bitcoin-client.html#installation-1>]

### Environment
 - Hardware platform:  Raspberry Pi 4
 - Operating system: Raspberry Pi OS 64bit lite (legacy)
 - Version: Debian GNU/Linux 11 (bullseye)

### Additional context
I formatted the SSD on a windows machine with one partition as an NTFS  before using the pi imager to write Raspberry Pi OS (Legacy, 64 bit) Lite

Is NTFS the correct format?
Should i have chosen the non legacy version of the OS?

Would appreciate your help.

'lsblk -pli' ouput:

> lNAME      MAJ:MIN RM  SIZE RO TYPE MOUNTPOINT
> /dev/sda    8:0    0  3.6T  0 disk 
> /dev/sda1   8:1    0  256M  0 part /boot
> /dev/sda2   8:2    0    4G  0 part /

'df -h' output:

> Filesystem      Size  Used Avail Use% Mounted on
> /dev/root       3.9G  3.7G   18M 100% /
> devtmpfs        3.6G     0  3.6G   0% /dev
> tmpfs           3.9G     0  3.9G   0% /dev/shm
> tmpfs           1.6G  1.2M  1.6G   1% /run
> tmpfs           5.0M  4.0K  5.0M   1% /run/lock
> /dev/sda1       255M   31M  225M  13% /boot
> tmpfs           782M   20K  782M   1% /run/user/1000

the error after 'tar -xvf bitcoin-$VERSION-aarch64-linux-gnu.tar.gz' command:
> bitcoin-29.0/
> bitcoin-29.0/README.md
> bitcoin-29.0/bin/
> bitcoin-29.0/bin/bitcoin-cli
> bitcoin-29.0/bin/bitcoin-qt
> tar: bitcoin-29.0/bin/bitcoin-qt: Wrote only 512 of 10240 bytes
> bitcoin-29.0/bin/bitcoin-tx
> tar: bitcoin-29.0/bin/bitcoin-tx: Cannot write: No space left on device
> bitcoin-29.0/bin/bitcoin-util
> tar: bitcoin-29.0/bin/bitcoin-util: Cannot write: No space left on device

kristapsk, 2025-04-24:
> /dev/root 3.9G 3.7G 18M 100% /

Here's the answer. Your physical SSD might be 4 TB, but your root partition is only 4 GB.

pleb4868, 2025-04-26:
It seems to be an issue with the pi imager.
No matter how i format and partition the drive before after the image burn I have 256 mb boot, ~1.7G root and the rest in 2 unallocated partitions (that seems to be a seperate issue where the imager is mbr which means it works with 2TB max)

Anyone else had a similar issue?

ciabutas, 2025-06-19:
you need use gparted to enlarge that partition