raspibolt/raspibolt #1487: [BUG] Incorrect permissions when verifying the bitcoin cookie

martinatime, 2025-06-02:
### Describe the bug
In https://raspibolt.org/guide/bitcoin/bitcoin-client.html#verification-of-bitcoind-operations there is a section to verify the bitcoin permission cookie

> $ ls -la /home/bitcoin/.bitcoin/.cookie
> -rw-r----- 1 bitcoin bitcoin 75 Dec 17 13:48 /home/bitcoin/.bitcoin/.cookie

The problem is that as "admin" you don't have access to "/home/bitcoin" (unless you use sudo) and therefore this command errors out with "Permission denied". This isn't a huge issue but a minor one (I may fix it myself).

### Location in guide
[Link to the corresponding page and section on <https://raspibolt.org>](https://raspibolt.org/guide/bitcoin/bitcoin-client.html#verification-of-bitcoind-operations)

### Expected behavior
Either add sudo to the command OR change the path to /home/admin/.bitcoin/.cookie

### Screenshots
N/A

### Environment
 N/A

### Additional context
N/A

martinatime, 2025-06-02:
Same issue exists when tailing the bitcoin debug.log
tail -f /home/bitcoin/.bitcoin/debug.log

martinatime, 2025-06-03:
admin@[redacted]:/home $ cd bitcoin/
-bash: cd: bitcoin/: Permission denied

blckbx, 2025-06-04:
okay this is odd because after adding a new test user:
```
admin@node:~$ sudo adduser --gecos "" --disabled-password test
```
for me, the home directory of that user has group read and access `x` by default:
```
admin@node:~$ ll /home/

drwxr-xr-x 18 root           root           4096 Jun  4 16:17 ./
drwxr-xr-x 27 root           root           4096 Aug 25  2024 ../
drwxr-x--- 16 admin          admin          4096 Jun  3 10:40 admin/
drwxr-x--- 14 bitcoin        bitcoin        4096 May 23 22:23 bitcoin/
[...]
drwxr-x---  2 test           test           4096 Jun  4 16:17 test/
```

martinatime, 2025-06-05:
Yeah, you have the /home/bitcoin/ directory with group permissions but that is not the default unless there is a step I missed to change that.

blckbx, 2025-07-24:
@BillyMabe456 
The `chmod` line in bitcoind.service has been removed in favor of the bitcoin conf setting: `rpccookieperms=group`

martinatime, 2025-09-22:
> [@BillyMabe456](https://github.com/BillyMabe456) The `chmod` line in bitcoind.service has been removed in favor of the bitcoin conf setting: `rpccookieperms=group`

Ok, does this mean the guide needs to be changed to reflect that or has it already been changed and I overlooked the change?

blckbx, 2025-09-22:
> > [@BillyMabe456](https://github.com/BillyMabe456) The `chmod` line in bitcoind.service has been removed in favor of the bitcoin conf setting: `rpccookieperms=group`
> 
> Ok, does this mean the guide needs to be changed to reflect that or has it already been changed and I overlooked the change?

It has been changed some time ago: https://github.com/raspibolt/raspibolt/commit/163bacdb8caaf63945ac9e93b0f278d981d9884b