raspibolt/raspibolt #1491: Bitcoin upgrade 29.0 instructions

luksocculta, 2025-06-18:
Upgrading Bitcoind to 29.0 causing Electrs fail to connect with:
0: bitcoind RPC polling failed
1: daemon not available
2: JSON error: invalid type: sequence, expected a string

This is done on test machine with testnet as I was afraid to make this upgrade on my RPI.
So. To solve this issue I was forced to upgrade to Bookworm, as newest version of Electrs is finally able to connect to Bitcond 29.0. (Probably some older too, but I installed newest directly from 0.9.14)

I made also some test and first version of Bitcoin where Electrs didnt to connect was 28.0.

So, has anybody tried this upgrade to 29.0 on real rpi with _bullseye_ and it works and I missing something or this instructions are forced quite too fast?

martinatime, 2025-06-19:
I think Bitcoin v29 is the cause of issues. Since I updated I was having issues with LND where LND has a healthcheck calling an RPC call against Bitcoin that was timing out. I had to disable the health check as when it fails three times LND restarts. Sometimes this happened multiple restarts in a row which caused a bunch of unstability. I'm still trying different things to see if I can resolve the underlying issue (Bitcoin responding very slowly to some RPC calls).

luksocculta, 2025-06-21:
i am sure that it happens with 28.0 too. i think there is no sense to make upstream issue as this is fixed on at least newest electrs and 9.14 is outdated anyway. with some free time i will think what to do. stay on bitcoin 26 as this serious bug is not affecting us as we run bitcoin on tor and wait for official move of raspibolt to newest debian, or make upgrade on my own.

time-out, 2025-08-14:
> Upgrading Bitcoind to 29.0 causing Electrs fail to connect with:
> 0: bitcoind RPC polling failed 1: daemon not available 2: JSON error: invalid type: sequence, expected a string

Same problem here with Bitcoin Knots v28.1.knots20250305 and Bitcoin Core v29. Now I'm back on v26.1 again. Electrs 0.9.14 does not work with v29 and _bullseye_ as OS, which is described in Raspibolt.

AsyncLuck, 2025-09-05:
Tried that: https://github.com/raspibolt/raspibolt/issues/1415 to update electrs and it works.

AsyncLuck, 2025-09-06:
@time-out you can even go with electrs 0.10.10 if needed. I was stuck on the old vs for a long time too but because I upgraded the node ssd to 2TB, I searched a little bit more to be able to upgrade all my things ;)